ZishuoYang/UT-Backplane-mapping #11: Parse Altium-exported netlist

ypsun-umd, 2018-08-20:
So far we prefer `pCad` netlist, since it looks like a `lisp` program, which is very easy to parse. Tom was suggesting some alternative format, the name which I forget.

ypsun-umd, 2018-08-21:
We discovered that we can leverage `pyparsing` library. The following snippet might be useful:

## File to be parsed:
```
(asciiHeader
  (asciiVersion 3 0)
  (fileUnits Mil)
)
```

## Code snippet
```python
import pyparsing as pp
parser = pp.nextedExpr()
parser.parseFile(<filename>)
```

ZishuoYang, 2018-08-29:
Add PcadReader to pyUTM/io.py for reading `Pcad` format netlist

ypsun-umd, 2018-08-31:
It seems that `v0.4` can do PT-side checking now. The remaining task is to do the DCB-sdie checking.

ypsun-umd, 2018-09-19:
DCB-DCB parsing has been implemented.

closed 2018-09-19 by ypsun-umd