ZishuoYang/UT-Backplane-mapping #18: Rules that need to be changed

ypsun-umd, 2018-09-07:
- `THERMISTOR_N` in Tom's list is connected to `AGND` (need to change our source code).
- `SDA` and `SCL` are flipped. (need to change in Excel).

ZishuoYang, 2018-09-17:
Warning: Altium's "copy as text" function does not preserve the order of a group of Signal texts. This could lead to assigning wrong pins.

ZishuoYang, 2018-09-18:
## Some inconsistencies found by SchematicCheck.py that do not concern rule-changing:
- [ ] _n_ `P_LV_RETURN` get tied together on JPU/JPL, resulting in _1_ signal name chosen as net name and _(n-1)_ signal names inconsistent with net name. 404 out of 1075 warnings seem to be of this type.

ypsun-umd, 2018-09-19:
Renewed connection checking list
```========DCB-PT========
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SCL_5_P, Zishuo: JD4_JP0_EC_HYB_i2C_SCL_5, NODE: DCB: JD4, DCB_PIN: H50, PT: JP0, PT_PIN: A7
NETNAME inconsistent: Tom's: JD4_JP0_EC_RESET_GPIO_5_P, Zishuo: JD4_JP0_EC_RESET_GPIO_5, NODE: DCB: JD4, DCB_PIN: H23, PT: JP0, PT_PIN: A10
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SCL_4_P, Zishuo: JD4_JP0_EC_HYB_i2C_SCL_4, NODE: DCB: JD4, DCB_PIN: H26, PT: JP0, PT_PIN: A17
NETNAME inconsistent: Tom's: JD4_JP0_EC_RESET_GPIO_3_P, Zishuo: JD4_JP0_EC_RESET_GPIO_3, NODE: DCB: JD4, DCB_PIN: H15, PT: JP0, PT_PIN: A31
NETNAME inconsistent: Tom's: JD4_JP0_THERM_EC_ADC_7_P, Zishuo: JD4_JP0_EC_ADC_7, NODE: DCB: JD4, DCB_PIN: H46, PT: JP0, PT_PIN: B5
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SDA_5_P, Zishuo: JD4_JP0_EC_HYB_i2C_SDA_5, NODE: DCB: JD4, DCB_PIN: H49, PT: JP0, PT_PIN: B8
NETNAME inconsistent: Tom's: JD4_JP0_EC_RESET_GPIO_4_P, Zishuo: JD4_JP0_EC_RESET_GPIO_4, NODE: DCB: JD4, DCB_PIN: H19, PT: JP0, PT_PIN: B21
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SDA_4_P, Zishuo: JD4_JP0_EC_HYB_i2C_SDA_4, NODE: DCB: JD4, DCB_PIN: H25, PT: JP0, PT_PIN: C17
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SDA_3_P, Zishuo: JD4_JP0_EC_HYB_i2C_SDA_3, NODE: DCB: JD4, DCB_PIN: H21, PT: JP0, PT_PIN: C28
NETNAME inconsistent: Tom's: JD4_JP0_THERM_EC_ADC_6_P, Zishuo: JD4_JP0_EC_ADC_6, NODE: DCB: JD4, DCB_PIN: H45, PT: JP0, PT_PIN: D14
NETNAME inconsistent: Tom's: JD4_JP0_EC_HYB_i2C_SCL_3_P, Zishuo: JD4_JP0_EC_HYB_i2C_SCL_3, NODE: DCB: JD4, DCB_PIN: H22, PT: JP0, PT_PIN: D28
NETNAME inconsistent: Tom's: JD0_JP0_EC_HYB_i2C_SCL_5_P, Zishuo: JD0_JP0_EC_HYB_i2C_SCL_5, NODE: DCB: JD0, DCB_PIN: H50, PT: JP0, PT_PIN: G26
NETNAME inconsistent: Tom's: JD2_JP0_EC_HYB_i2C_SCL_5_P, Zishuo: JD2_JP0_EC_HYB_i2C_SCL_5, NODE: DCB: JD2, DCB_PIN: H50, PT: JP0, PT_PIN: H8
NETNAME inconsistent: Tom's: JD2_JP0_EC_RESET_GPIO_5_P, Zishuo: JD2_JP0_EC_RESET_GPIO_5, NODE: DCB: JD2, DCB_PIN: H23, PT: JP0, PT_PIN: I3
NETNAME inconsistent: Tom's: JD2_JP0_EC_HYB_i2C_SDA_5_P, Zishuo: JD2_JP0_EC_HYB_i2C_SDA_5, NODE: DCB: JD2, DCB_PIN: H49, PT: JP0, PT_PIN: I9
NETNAME inconsistent: Tom's: JD0_JP0_EC_HYB_i2C_SDA_5_P, Zishuo: JD0_JP0_EC_HYB_i2C_SDA_5, NODE: DCB: JD0, DCB_PIN: H49, PT: JP0, PT_PIN: I26
NETNAME inconsistent: Tom's: JD0_JP0_EC_RESET_GPIO_5_P, Zishuo: JD0_JP0_EC_RESET_GPIO_5, NODE: DCB: JD0, DCB_PIN: H23, PT: JP0, PT_PIN: J22
NETNAME inconsistent: Tom's: JD0_JP0_THERM_EC_ADC_7_P, Zishuo: JD0_JP0_EC_ADC_7, NODE: DCB: JD0, DCB_PIN: H46, PT: JP0, PT_PIN: J27
NETNAME inconsistent: Tom's: JD0_JP1_EC_HYB_i2C_SCL_1_P, Zishuo: JD0_JP1_EC_HYB_i2C_SCL_1, NODE: DCB: JD0, DCB_PIN: H14, PT: JP1, PT_PIN: A7
NETNAME inconsistent: Tom's: JD0_JP1_EC_RESET_GPIO_1_P, Zishuo: JD0_JP1_EC_RESET_GPIO_1, NODE: DCB: JD0, DCB_PIN: H27, PT: JP1, PT_PIN: A10
NETNAME inconsistent: Tom's: JD4_JP1_EC_HYB_i2C_SCL_2_P, Zishuo: JD4_JP1_EC_HYB_i2C_SCL_2, NODE: DCB: JD4, DCB_PIN: H18, PT: JP1, PT_PIN: A17
NETNAME inconsistent: Tom's: JD4_JP1_EC_RESET_GPIO_1_P, Zishuo: JD4_JP1_EC_RESET_GPIO_1, NODE: DCB: JD4, DCB_PIN: H27, PT: JP1, PT_PIN: A31
NETNAME inconsistent: Tom's: JD0_JP1_EC_HYB_i2C_SDA_1_P, Zishuo: JD0_JP1_EC_HYB_i2C_SDA_1, NODE: DCB: JD0, DCB_PIN: H13, PT: JP1, PT_PIN: B8
NETNAME inconsistent: Tom's: JD4_JP1_EC_RESET_GPIO_2_P, Zishuo: JD4_JP1_EC_RESET_GPIO_2, NODE: DCB: JD4, DCB_PIN: H11, PT: JP1, PT_PIN: B21
NETNAME inconsistent: Tom's: JD2_JP1_THERM_EC_ADC_3_P, Zishuo: JD2_JP1_EC_ADC_3, NODE: DCB: JD2, DCB_PIN: H38, PT: JP1, PT_PIN: B39
NETNAME inconsistent: Tom's: JD4_JP1_EC_HYB_i2C_SDA_2_P, Zishuo: JD4_JP1_EC_HYB_i2C_SDA_2, NODE: DCB: JD4, DCB_PIN: H17, PT: JP1, PT_PIN: C17
NETNAME inconsistent: Tom's: JD4_JP1_THERM_EC_ADC_3_P, Zishuo: JD4_JP1_EC_ADC_3, NODE: DCB: JD4, DCB_PIN: H38, PT: JP1, PT_PIN: C24
NETNAME inconsistent: Tom's: JD4_JP1_EC_HYB_i2C_SDA_1_P, Zishuo: JD4_JP1_EC_HYB_i2C_SDA_1, NODE: DCB: JD4, DCB_PIN: H13, PT: JP1, PT_PIN: C28
NETNAME inconsistent: Tom's: JD4_JP1_THERM_EC_ADC_4_P, Zishuo: JD4_JP1_EC_ADC_4, NODE: DCB: JD4, DCB_PIN: H41, PT: JP1, PT_PIN: D14
NETNAME inconsistent: Tom's: JD4_JP1_EC_HYB_i2C_SCL_1_P, Zishuo: JD4_JP1_EC_HYB_i2C_SCL_1, NODE: DCB: JD4, DCB_PIN: H14, PT: JP1, PT_PIN: D28
NETNAME inconsistent: Tom's: JD2_JP1_EC_HYB_i2C_SDA_1_P, Zishuo: JD2_JP1_EC_HYB_i2C_SDA_1, NODE: DCB: JD2, DCB_PIN: H13, PT: JP1, PT_PIN: F34
NETNAME inconsistent: Tom's: JD2_JP1_EC_RESET_GPIO_1_P, Zishuo: JD2_JP1_EC_RESET_GPIO_1, NODE: DCB: JD2, DCB_PIN: H27, PT: JP1, PT_PIN: F39
NETNAME inconsistent: Tom's: JD0_JP1_EC_HYB_i2C_SCL_3_P, Zishuo: JD0_JP1_EC_HYB_i2C_SCL_3, NODE: DCB: JD0, DCB_PIN: H22, PT: JP1, PT_PIN: G26
NETNAME inconsistent: Tom's: JD2_JP1_EC_HYB_i2C_SCL_1_P, Zishuo: JD2_JP1_EC_HYB_i2C_SCL_1, NODE: DCB: JD2, DCB_PIN: H14, PT: JP1, PT_PIN: G33
NETNAME inconsistent: Tom's: JD2_JP1_EC_HYB_i2C_SCL_3_P, Zishuo: JD2_JP1_EC_HYB_i2C_SCL_3, NODE: DCB: JD2, DCB_PIN: H22, PT: JP1, PT_PIN: H8
NETNAME inconsistent: Tom's: JD2_JP1_EC_RESET_GPIO_3_P, Zishuo: JD2_JP1_EC_RESET_GPIO_3, NODE: DCB: JD2, DCB_PIN: H15, PT: JP1, PT_PIN: I3
NETNAME inconsistent: Tom's: JD2_JP1_EC_HYB_i2C_SDA_3_P, Zishuo: JD2_JP1_EC_HYB_i2C_SDA_3, NODE: DCB: JD2, DCB_PIN: H21, PT: JP1, PT_PIN: I9
NETNAME inconsistent: Tom's: JD0_JP1_EC_HYB_i2C_SDA_3_P, Zishuo: JD0_JP1_EC_HYB_i2C_SDA_3, NODE: DCB: JD0, DCB_PIN: H21, PT: JP1, PT_PIN: I26
NETNAME inconsistent: Tom's: JD0_JP1_EC_RESET_GPIO_3_P, Zishuo: JD0_JP1_EC_RESET_GPIO_3, NODE: DCB: JD0, DCB_PIN: H15, PT: JP1, PT_PIN: J22
NETNAME inconsistent: Tom's: JD0_JP1_THERM_EC_ADC_5_P, Zishuo: JD0_JP1_EC_ADC_5, NODE: DCB: JD0, DCB_PIN: H42, PT: JP1, PT_PIN: J27

```

ZishuoYang, 2019-02-07:
## New rules from Tom O's latest schematic editing (07 Feb 2019) to be implemented in NetlistGen and/or NetlistCheck: 
[Reference: [FullBP_Notes_TomO_07Feb2019.pdf](https://github.com/ZishuoYang/UT-Backplane-mapping/files/2842567/FullBP_Notes_TomO_07Feb2019.pdf)]

- [ ] ALL FRO (never-used) elink inputs must be connected to 1.5V / ground bias connections:
  - [x] Confirm this high impedance setting can be accomplished on each individual port rather than as part of a larger bank setting (ie becomes an issue if a bank needs to be set to 100 Ohm because some of the elinks are used, but some are FRO). See https://github.com/umd-lhcb/lab-notes/wiki/Terminate-each-elink-individually
  - [ ] THE FRO_P and FRO_N associated with EACH DCB slot should be connected to a COMMON 1.5V and ground via 100 Ohm series resistors. The FRO bias 1.5V source must be for the same DCB slot only. This should be done by changing `JD#_FRO_DC#_ELK_CH#_P`  -> `JD#_FRO_ELK_P` and `JD#_FRO_DC#_ELK_CH#_N`  -> `JD#_FRO_ELK_N` in JD pin assignments, and then connect `JD#_FRO_ELK_`P/N to R(100Ohm) to `JD#_..._1V5`/`GND`
  - [ ] ALSO applies to the unused MC and EC **DIN_ELKs** in each DCB slot, i.e. `JD#_FRO_MC_SEC_DIN_ELK` and `JD#_FRO_EC_SEC_DIN_ELK`

- [ ] Add ADC connections for the ground sense associated with the three most remote hybrid power groups P3, P2_EAST, P1_WEST

- [ ] TDB

yipengsun, 2019-02-08:
For `EC_ADC_REF13`, on the DCB yaml file, no `JD` connector is specified. How do we know that `JD0` should connect to `JP0`'s sense ground?

yipengsun, 2019-02-08:
OK, it seems that we should assume `JD#_EC_ADC_REFXX` should be connected to `JP#`'s sense ground.

tobannon, 2019-02-08:
The following inputs are associated with the unused secondary slow control connections.  These are tied to a common bias set separate from the other FRO inputs.  The separate components are required to accommodate the 100 terminations already installed on the DCB.  Dual parallel 100 ohm resistors are used for each leg to ensure there is at least a 350 to 400 mV differential voltage present.  The redundant 100 ohm components are used to simplify the assembly logistics by negating the need to add another component value to the assembly.
JD0_FRO_MC_SEC_DIN_ELK_N	--> JD0_FRO_SEC_ELK_N
JD0_FRO_MC_SEC_DIN_ELK_P	        --> JD0_FRO_SEC_ELK_P
JD0_FRO_EC_SEC_DIN_ELK_N	        --> JD0_FRO_SEC_ELK_P
JD0_FRO_EC_SEC_DIN_ELK_P	        --> JD0_FRO_SEC_ELK_N
JD0_FRO_EC_SEC_CLK_ELK_N	        --> JD0_FRO_SEC_ELK_P
JD0_FRO_EC_SEC_CLK_ELK_P	        --> JD0_FRO_SEC_ELK_N

ZishuoYang, 2019-02-11:
- [ ] For now, to keep error-checking going, we have to patch our code and bypass the inconsistent pinout in pigtail SEAM library part, #55.

CoffeeIntoScience, 2019-02-18:
>For each JD, add ADC connections for the ground sense associated with the three most remote hybrid power groups P3, P2_EAST, P1_WEST.

These are not power groups, these are hybrids. This rule in general seems ambiguous because a DCB is not uniquely associated with a pigtail. Please clarify.

ZishuoYang, 2019-02-19:
As described in [Reference: [FullBP_Notes_TomO_07Feb2019.pdf](https://github.com/ZishuoYang/UT-Backplane-mapping/files/2842567/FullBP_Notes_TomO_07Feb2019.pdf)] by @tobannon , this is added input to ADC for measuring hybrids that are physically present in all variants (ie P1_WEST, P2_EAST, & P3). 
He wanted each DCB to associate with a PT, i.e. JD0-JP0 etc. This raises a good point: if JD2-JP2, then on beta JP2 won't be have JD2 to connect to.

ZishuoYang, 2019-02-22:
**Review comments from Phoebe's, Feb 18 2019:**

> Regarding Backplane Error Checking. I am trying to think of any corner cases that could be missed based on the list Zishuo has shown me.
> - Suppose a line appeared twice from a copy-paste error or (heaven forbid) all the way back in the original map: would it have been caught? (inverse of #4 on issue 43 essentially)
> - Is there any line its possible to have dropped? Say we go from ELK_CH11 - ELK_CH10 - ELK_CH8 stepping down. Does your check that the channel number decreases find if it skips one?
> - Has any checking been done that every sense line finds its way to the telemetry connector one way or another? Is it possible to do so?
> - Similar for the LV lines. Can we be sure that every one makes its way off the backplane either at one of these godawful swage connectors or to JPLx?
> 
> - I also put a couple clarifying questions on the issues themselves to make sure I understand what's been done
> - It would be nice to also have a way to know for sure that all signals with a _P or _N received a label of DIFFPAIR or SE_DIFFPAIR, but it's not obvious this is possible

closed 2020-12-10 by yipengsun