ZishuoYang/UT-Backplane-mapping #22: TODO: update mirror mapping Excel docs regarding SCL/SDA

ZishuoYang, 2018-09-18:
v5.2 has been issued for trueType mapping. The same needs to be done on the mirrorType, to correct the error made on DCB boards.

ypsun-umd, 2018-10-02:
Now we probably need to make changes when generating Excel from yaml definition.

yipengsun, 2018-12-07:
With current state, this is not needed, as we now have a prototype mapping (those `.yml` files), which is neither True-type nor Mirror-type. The real mappings will be generated from the prototype mapping.

ZishuoYang, 2019-02-19:
I've spot-checked that SCL and SDA have been swapped on both the current .yml files and the v5.2 Excel docs. Just need to make sure that @tobannon is aware that this change is applied to both the Pathfinder and the Full.

closed 2019-02-19 by ZishuoYang