ZishuoYang/UT-Backplane-mapping #29: Possible problem: no parenthesis after "or"

ypsun-umd, 2018-09-26:
In the latest commit:
```python
...for pt_id in range(0, len(pt_descr)):
    for pt_entry in pt_descr[pt_id]:
        if pt_entry['Signal ID'] is not None \
                and (pt_entry['Signal ID'].endswith('SCL_N') or
                     pt_entry['Signal ID'].endswith('SDA_N') or
                     pt_entry['Signal ID'].endswith('RESET_N')):
```
There is no parenthesis after the second `and`. These `or`s might not be behave as intended.

ypsun-umd, 2018-09-26:
On a second thought, that snippet seems to be correct, since `and` binds tighter than `or`, once the first 2 conditions are satisfied, `Python` would give you `True`.

Still, I'd like to confirm that this is the intended behavior.

ZishuoYang, 2018-09-26:
Yes, the intended behaviors is to return `True` when any of the three OR conditions is `True`

closed 2018-09-26 by ZishuoYang