ZishuoYang/UT-Backplane-mapping #43: Additional error-checking of pin assignments

ZishuoYang, 2018-12-07:
Two categories of possible errors can be checked, as was done for the Excel docs:

- **Errors due to pin-assigning in the process of mapping each net/connection:**
1. Check that polarity is correct: _N/P of one pin matches _N/P of the other pin.
2. Check if the mapped pins have the same type of signal ID: e.g. Pigtail's signal ID (ASIC_CH) agrees with DCB’s signal ID (DC_ELK).
3. Check that connector pins in the two list match.
4. Check if a specific pin on a specific connector is used more than once.

- **Errors due to violating design rules, for each type of signals:**
1. Check the Elink Packing Order between EC_ELK and ASIC_CH
2. Check MC_OUT_RCLK to DC_IN_RCLK
3. Check MC_SEC to EC_SEC
4. Check MC_TFC to Hybrids and the number of signals
5. Check DC_OUT_RCLK (from DCs to Hybrids)
6. Check EC_RESET to DC and Hybrids and the number of signals
7. Check EC_HYB_i2C (from SCA to Hybrids)
8. Check EC_ADC to Hybrid thermistors

CoffeeIntoScience, 2019-02-18:
When you say, for example, that "MC_TFC connects to hybrids and the number of signals" does that mean you've checked that every 4-asic group gets one TFC assigned?

ZishuoYang, 2019-02-19:
No it's more basic than that. At this stage I was more concerned with whether a signal has been assigned to the right connector. So the checks regarding the control signals (MC, EC, etc) were checking for exmaple: if a TFC signal is from JD9/JD10 does it go to JP8/9/10/11? And count the number of TFC signals going to certain JPs and check that matches the way I assigned them. 

I'll update the descriptions above.

ZishuoYang, 2019-02-19:
> does that mean you've checked that every 4-asic group gets one TFC assigned?
The way mapping was done made sure every 4-asic group gets one TFC assigned. It's also been checked by eye,

ZishuoYang, 2019-02-22:
**Review comments from Phoebe, Feb 18 2019. With my replies followed.**

> Regarding Backplane Error Checking. I am trying to think of any corner cases that could be missed based on the list Zishuo has shown me.
> - Suppose a line appeared twice from a copy-paste error or (heaven forbid) all the way back in the original map: would it have been caught? (inverse of item 4 on #43 essentially)
> - Is there any line its possible to have dropped? Say we go from ELK_CH11 - ELK_CH10 - ELK_CH8 stepping down. Does your check that the channel number decreases find if it skips one?

*I have checked that within a subframe block (see figure below) the numbers do not skip. But we can check additionally between blocks. -ZSY*
[image]

> - Has any checking been done that every sense line finds its way to the telemetry connector one way or another? Is it possible to do so?
> - Similar for the LV lines. Can we be sure that every one makes its way off the backplane either at one of these godawful swage connectors or to JPLx?
> 
> - I also put a couple clarifying questions on the issues themselves to make sure I understand what's been done

*These have been answered. -ZSY*

> - It would be nice to also have a way to know for sure that all signals with a _P or _N received a label of DIFFPAIR or SE_DIFFPAIR, but it's not obvious this is possible

*Possible, but need to add function in NetlistCheck.py for parsing the netlist to get this additional label.- ZSY*

ZishuoYang, 2019-02-22:
We have implemented the following general checks:

- Check that all the netnames from the pin-assignment copy-paste lists are present on the netlist generated from the schematics.

ZishuoYang, 2019-02-22:
For @yipengsun to develop additional checking on the single-to-differential signals:
[image]