ZishuoYang/UT-Backplane-mapping #55: Nets Being questioned

tobannon, 2019-01-18:
[Nets_Being_Questioned_FullTrueBP_Jan18_2019.pdf](https://github.com/ZishuoYang/UT-Backplane-mapping/files/2774497/Nets_Being_Questioned_FullTrueBP_Jan18_2019.pdf)

CoffeeIntoScience, 2019-01-18:
Pins JP0_d29 and JP1_b29 are both 'unused float' and should be tied to GND

CoffeeIntoScience, 2019-01-18:
@ZishuoYang @yipengsun for future versions of the mapping "ForRefOnly" should be "FRO" and "_AGND" should be "AGND" for consistency with the pathfinder notation

yipengsun, 2019-01-22:
For the `*_AGND`, that has been fixed in release 0.6.1 a couple of days ago. We noticed that problem and fixed that in our generating rules.

For `ForRefOnly`, that is an easy string replacement.

ZishuoYang, 2019-01-22:
The updated NetlistCheck.py has picked up nets `JD0_JP1_MC_TFC1_N` `JD0_JP1_MC_TFC1_P` as 'not implemented':
>* NOT implemented: NET: JD0_JP1_MC_TFC1_N, NODE: DCB: JD0, DCB_PIN: F1, PT: JP1, PT_PIN: A9
NOT implemented: NET: JD0_JP1_MC_TFC1_P, NODE: DCB: JD0, DCB_PIN: F2, PT: JP1, PT_PIN: B9

tobannon, 2019-01-24:
List of pin-assignment corrections implemented:
  JP0, D29: GND				---- ‘WAS’ = NOT CONNECTED
  JP0, B29: GND    				----**NO ISSUE FOUND (ie already grounded)**
  JD0, F1: JD0_JP1_MC_TFC1_N		---- ‘WAS’ = JD0_JP1C_TFC1_N
  JD0, F2: JD0_JP1_MC_TFC1_P		---- ‘WAS’ = JD0_JP1C_TFC1_P
  JD0, H38: JD0_JP1_THERM_EC_ADC_3_P	---- ‘WAS’ = JD0_FRO_EC_ADC_3
  JD10, E5: GND				---- ‘WAS’ = NOT CONNECTED
  JD0, B1: JD0_FRO_MC_TFC0_N		---- ‘WAS’ = JD0_FROC_TFC0_N
  JD0, B2: JD0_FRO_MC_TFC0_P		---- ‘WAS’ = JD0_FROC_TFC0_p
  R403-1:  Delete				---- ‘WAS’ = deleted JP2_JT0_P2_EAST_THERM_N & R403
  R404-1:  Delete				---- ‘WAS’ = deleted JP2_JT0_P2_EAST_THERM_P & R404
  Search for all JD#AGND and replace with JD#_AGND (# is from 0 to 11. Expect to change only JP pages )
DONE:  ‘WAS’ = JD0AGND, JD1AGND, JD3AGND, JD4AGND, JD5AGND, JD6AGND, JD7AGND, JD8AGND, JD9AGND, JD10AGND, JD11AGND
NOT FOUND:  JD2AGND (JD2_AGND already done)

NEW ISSUE found is JD11GND is connected only to NT50.  Presume this should be JD11_AGND ----PLEASE CONFIRM

ZishuoYang, 2019-02-08:
WIP #18, NetlistCheck picked up inconsistent net name for: 
- `JP2 J10`, `GND`!= `JP2_JPU0_P1_EAST_LV_SOURCE`

ZishuoYang, 2019-02-11:
- [ ] A minor inconsistency was found for DCB remote reset netname, where on schematic there is a space between `JD#_REMOTE` and `_RESETB`. @tobannon should update the schematic to avoid space within netnames.

ZishuoYang, 2019-02-13:
- [ ] WIP #18, found incorrect designators for depopulation resistors. To be changed to `RB_###`:
[image]

ZishuoYang, 2019-02-14:
- [ ] WIP #18, found FRO ELKs that are not properly biased. The signal types include `JD#_FRO_MC_SEC_DIN_ELK` and `JD#_FRO_EC_SEC_DIN_ELK`, e.g. 

    [image]

ZishuoYang, 2019-02-22:
While this is not regarding nets, I include the following check I did for the schematics here for documentation purpose:
[image]
The SEAM pin designators and connector orientations shown in the picture have been checked/updated.

ZishuoYang, 2019-02-26:
WIP #18, found a minor error in designators of biasing resistors for FRO ELKs. Need to update ~8 resistor names from `RB_#` to `R_#`. (Because these resistors are for FRO ELKs, they should be on all variants. @tobannon )

[image]

> Checking never-used FRO ELKs...
> No biasing resistor found in JD9_FRO_ELK_P
> No biasing resistor found in JD9_FRO_ELK_N
> No biasing resistor found in JD8_FRO_ELK_P
> No biasing resistor found in JD8_FRO_ELK_N
> No biasing resistor found in JD7_FRO_ELK_P
> No biasing resistor found in JD7_FRO_ELK_N
> No biasing resistor found in JD6_FRO_ELK_P
> No biasing resistor found in JD6_FRO_ELK_N
> No biasing resistor found in JD4_FRO_ELK_P
> No biasing resistor found in JD4_FRO_ELK_N
> No biasing resistor found in JD3_FRO_ELK_P
> No biasing resistor found in JD3_FRO_ELK_N
> No biasing resistor found in JD2_FRO_ELK_P
> No biasing resistor found in JD2_FRO_ELK_N
> No biasing resistor found in JD0_FRO_ELK_P
> No biasing resistor found in JD0_FRO_ELK_N