ZishuoYang/UT-Backplane-mapping #65: to include control signals in the UT-system ASIC-Fiber mapping

ZishuoYang, 2019-02-15:
As titled, to extend the current fiber mapping header and include:
- DCB channel for `DC_OUT_RCLK#_P` (for JP signal e.g. `P#_EAST_CLK_P`)
- DCB channel for `EC_HYB_i2C_SCL_#` (for JP signal e.g. `P#_EAST_I2C_SCL_P`)
- DCB channel for `EC_HYB_i2C_SDA_#` (for JP signal e.g. `P#_EAST_I2C_SDA_P`)
- DCB channel for `EC_RESET_GPIO_#` (for JP signal e.g. `P#_EAST_RESET_P`)
- DCB channel for `MC_TFC#_P` (for JP signal e.g. `P#_EAST_TFC_P`)
- DCB channel for `EC_ADC_#` (for JP signal e.g. `P#_EAST_THERMISTOR_P`)

*Note that each set of these channels is for one 4-ASIC group (e.g. P1_EAST).*

yipengsun, 2019-02-15:
So far, when viewing the `ctrls_descr_alpha`, we have:
```
>>> ctrls_descr_alpha['X-0-M']['P1_WEST_ASIC_3']
[{'hybrid': 'P1', 'dcb_idx': 2, 'ctrl_signal_type': 'DC_OUT_RCLK5_P'}, {'hybrid': 'P1', 'dcb_idx': 0, 'ctrl_signal_type': 'EC_HYB_i2C_SCL_5'}, {'hybrid': 'P1', 'dcb_idx': 2, 'ctrl_signal_type': 'EC_HYB_i2C_SCL_5'}, {'hybrid': 'P1', 'dcb_idx': 0, 'ctrl_signal_type': 'DC_OUT_RCLK5_P'}, {'hybrid': 'P1', 'dcb_idx': 2, 'ctrl_signal_type': 'EC_RESET_GPIO_5'}, {'hybrid': 'P1', 'dcb_idx': 2, 'ctrl_signal_type': 'EC_HYB_i2C_SDA_5'}, {'hybrid': 'P1', 'dcb_idx': 0, 'ctrl_signal_type': 'EC_HYB_i2C_SDA_5'}, {'hybrid': 'P1', 'dcb_idx': 2, 'ctrl_signal_type': 'MC_TFC5_P'}, {'hybrid': 'P1', 'dcb_idx': 0, 'ctrl_signal_type': 'EC_RESET_GPIO_5'}, {'hybrid': 'P1', 'dcb_idx': 0, 'ctrl_signal_type': 'MC_TFC5_P'}]
>>> ctrls_descr_alpha['X-0-M']['P1_WEST_ASIC_3']
```

ZishuoYang, 2019-02-17:
Near complete. TODO: deal with the exceptional thermistors rerouted to telemetry breakout boards (JT)

yipengsun, 2019-02-19:
For these `EC_ADC` channels, we need to update the `AltiumNetlistGen.py` so that they have correct netnames, because they should be connected to `JT`, instead of `GND`.

ZishuoYang, 2019-02-19:
After spot-checking, consider this fully implemented.

closed 2019-02-19 by ZishuoYang