ZishuoYang/UT-Backplane-mapping #72: Possible error on checking depopulation components

yipengsun, 2019-02-23:
Currently:
```

    if True in map(lambda x: bool(re.search(r'^JP8|^JP9|^JP10|^JP11', x[0])), components):
        if True not in map(lambda x: bool(re.search(r'^RB_\d+|^RBSP\d+|^RxCB_\d+', x[0])), components):
            print("No depopulation component found in {}".format(diff_net))
    # for Beta variant:
    else:
        if True not in map(lambda x: bool(re.search(r'^RB_\d+|^RBSP\d+|^RxCB_\d+', x[0])), components):
            print("No depopulation component found in {}".format(diff_net))
```

The `re.search` part is the same for both beta and beta special.

ZishuoYang, 2019-02-24:
Thanks for finding this! I need to think more about this but probably for beta it should be `re.search(r'^RB_\d+|^RxCB_\d+', x[0])` instead.

ZishuoYang, 2019-02-25:
No change in error log after fixing the bug. Will close this after committing fix.

commit 1550b9a8 (2019-02-25): Fix minot bug. Close #72.
commit d3b6e27c (2019-04-04): Fix #72: Don't check Beta SP components when dealing with Beta variant.

Also fixed various logical bugs.
closed 2019-02-25 by ZishuoYang