ZishuoYang/UT-Backplane-mapping #80: need to fix breakout-board signals (i.e. JT) on Mirror-type netlist gen/check

ZishuoYang, 2019-04-30:
Currently the netlist check reports the following:
>The following net is missing in the implementation: JP0_JT1_P3_LV_SENSE_P
The following net is missing in the implementation: JP0_JT1_P4_LV_SENSE_P
The following net is missing in the implementation: JP0_JT1_P4_LV_SENSE_N
The following net is missing in the implementation: JP0_JT1_P3_LV_SENSE_N
The following net is missing in the implementation: JP2_JT0_P3_LV_SENSE_P
The following net is missing in the implementation: JP2_JT0_P4_LV_SENSE_P
The following net is missing in the implementation: JP2_JT0_P4_LV_SENSE_N
The following net is missing in the implementation: JP2_JT0_P3_LV_SENSE_N>

This should not happen because we expect JT0 to associate with JP0-3.

ZishuoYang, 2019-05-01:
The second type of inconsistencies reported is because that biasing resistor for `FRO_ELK` should have designator `R#` not `RB_#`. These are actual error on the current schematics:
>No biasing resistor found in JD9_FRO_ELK_P
No biasing resistor found in JD9_FRO_ELK_N
No biasing resistor found in JD8_FRO_ELK_P
No biasing resistor found in JD8_FRO_ELK_N
No biasing resistor found in JD7_FRO_ELK_P
No biasing resistor found in JD7_FRO_ELK_N
No biasing resistor found in JD6_FRO_ELK_P
No biasing resistor found in JD6_FRO_ELK_N
No biasing resistor found in JD4_FRO_ELK_P
No biasing resistor found in JD4_FRO_ELK_N
No biasing resistor found in JD3_FRO_ELK_P
No biasing resistor found in JD3_FRO_ELK_N
No biasing resistor found in JD2_FRO_ELK_P
No biasing resistor found in JD2_FRO_ELK_N
No biasing resistor found in JD0_FRO_ELK_P
No biasing resistor found in JD0_FRO_ELK_N
>

ZishuoYang, 2019-05-01:
Reported a net named `JP11_JPL2_P3_LV_SOURCE` tied to a P2B2 socket `JS_PT_NINE_TEN_1`. This is suspicious because it's not used anywhere and there already exits `JP11_JPU2_P3_LV_SOURCE`.

yipengsun, 2019-05-01:
The full error log for `JPUX` and `JPLX`:
```
The following components are missing in the expected net JP11_JPL2_P3_LV_SOURCE: JP11-B15, JP11-B16, JP11-C15, JP11-C16, JP11-D15, JP11-D16, JP11-E15, JP11-E16, JP11-E17, JP11-F15, JP11-F16, JP11-F17
The following components are missing in the expected net JP11_JPL2_P3_LV_RETURN: JP11-A19, JP11-A20, JP11-A21, JP11-A22, JP11-E19, JP11-E20, JP11-E21, JP11-E22, JP11-F19, JP11-F20
The following components are missing in the expected net JP9_JPU2_P3_LV_SOURCE: JP9-B15, JP9-B16, JP9-C15, JP9-C16, JP9-D15, JP9-D16, JP9-E15, JP9-E16, JP9-E17, JP9-F15, JP9-F16, JP9-F17
The following components are missing in the expected net JP9_JPU2_P3_LV_RETURN: JP9-A19, JP9-A20, JP9-A21, JP9-A22, JP9-E19, JP9-E20, JP9-E21, JP9-E22, JP9-F19, JP9-F20
The following components are missing in the expected net JP7_JPU2_P3_LV_SOURCE: JP7-B15, JP7-B16, JP7-C15, JP7-C16, JP7-D15, JP7-D16, JP7-E15, JP7-E16, JP7-E17, JP7-F15, JP7-F16, JP7-F17
The following components are missing in the expected net JP7_JPU2_P3_LV_RETURN: JP7-A19, JP7-A20, JP7-A21, JP7-A22, JP7-E19, JP7-E20, JP7-E21, JP7-E22, JP7-F19, JP7-F20
The following components are missing in the expected net JP7_JPU2_P2_EAST_LV_SOURCE: JP7-E1, JP7-E2, JP7-E3, JP7-E4, JP7-E5, JP7-E6, JP7-E7
The following components are missing in the expected net JP7_JPU2_P2_EAST_LV_RETURN: JP7-A6, JP7-B6, JP7-C6, JP7-C7, JP7-D6, JP7-D7, JP7-E11
The following components are missing in the expected net JP7_JPU2_P1_WEST_LV_SOURCE: JP7-F27, JP7-F28, JP7-F29, JP7-F30, JP7-F31, JP7-F32, JP7-F33, JP7-H33, JP7-I33, JP7-J33
The following components are missing in the expected net JP7_JPU2_P1_WEST_LV_RETURN: JP7-F21, JP7-F22, JP7-F23, JP7-F24, JP7-F25, JP7-F26, JP7-G24, JP7-G25, JP7-H24, JP7-H25, JP7-I24
The following components are missing in the expected net JP7_JPU2_P1_EAST_LV_SOURCE: JP7-F11, JP7-F12, JP7-F13, JP7-F14, JP7-G10, JP7-G11, JP7-H10, JP7-H11, JP7-I11, JP7-J10, JP7-J11
The following components are missing in the expected net JP7_JPU2_P1_EAST_LV_RETURN: JP7-G5, JP7-G6, JP7-G7, JP7-G8, JP7-G9, JP7-H5, JP7-H6, JP7-H7, JP7-I7
The following components are missing in the expected net JP5_JPU1_P3_LV_SOURCE: JP5-B15, JP5-B16, JP5-C15, JP5-C16, JP5-D15, JP5-D16, JP5-E15, JP5-E16, JP5-E17, JP5-F15, JP5-F16, JP5-F17
The following components are missing in the expected net JP5_JPU1_P3_LV_RETURN: JP5-A19, JP5-A20, JP5-A21, JP5-A22, JP5-E19, JP5-E20, JP5-E21, JP5-E22, JP5-F19, JP5-F20
The following components are missing in the expected net JP5_JPU1_P2_EAST_LV_SOURCE: JP5-E1, JP5-E2, JP5-E3, JP5-E4, JP5-E5, JP5-E6, JP5-E7
The following components are missing in the expected net JP5_JPU1_P2_EAST_LV_RETURN: JP5-A6, JP5-B6, JP5-C6, JP5-C7, JP5-D6, JP5-D7, JP5-E11
The following components are missing in the expected net JP5_JPU1_P1_WEST_LV_SOURCE: JP5-F27, JP5-F28, JP5-F29, JP5-F30, JP5-F31, JP5-F32, JP5-F33, JP5-H33, JP5-I33, JP5-J33
The following components are missing in the expected net JP5_JPU1_P1_WEST_LV_RETURN: JP5-F21, JP5-F22, JP5-F23, JP5-F24, JP5-F25, JP5-F26, JP5-G24, JP5-G25, JP5-H24, JP5-H25, JP5-I24
The following components are missing in the expected net JP5_JPU1_P1_EAST_LV_SOURCE: JP5-F11, JP5-F12, JP5-F13, JP5-F14, JP5-G10, JP5-G11, JP5-H10, JP5-H11, JP5-I11, JP5-J10, JP5-J11
The following components are missing in the expected net JP5_JPU1_P1_EAST_LV_RETURN: JP5-G5, JP5-G6, JP5-G7, JP5-G8, JP5-G9, JP5-H5, JP5-H6, JP5-H7, JP5-I7
The following components are missing in the expected net JP3_JPU1_P3_LV_SOURCE: JP3-B15, JP3-B16, JP3-C15, JP3-C16, JP3-D15, JP3-D16, JP3-E15, JP3-E16, JP3-E17, JP3-F15, JP3-F16, JP3-F17
The following components are missing in the expected net JP3_JPU1_P3_LV_RETURN: JP3-A19, JP3-A20, JP3-A21, JP3-A22, JP3-E19, JP3-E20, JP3-E21, JP3-E22, JP3-F19, JP3-F20
The following components are missing in the expected net JP3_JPU1_P2_EAST_LV_SOURCE: JP3-E1, JP3-E2, JP3-E3, JP3-E4, JP3-E5, JP3-E6, JP3-E7
The following components are missing in the expected net JP3_JPU1_P2_EAST_LV_RETURN: JP3-A6, JP3-B6, JP3-C6, JP3-C7, JP3-D6, JP3-D7, JP3-E11
The following components are missing in the expected net JP3_JPL1_P4_LV_SOURCE: JP3-A23, JP3-A24, JP3-A25, JP3-A26, JP3-B27, JP3-C27, JP3-D27, JP3-E27
The following components are missing in the expected net JP3_JPL1_P4_LV_RETURN: JP3-A30, JP3-B30, JP3-E30, JP3-E31, JP3-E32, JP3-E33
The following components are missing in the expected net JP2_JPU1_P3_LV_SOURCE: JP2-B15, JP2-B16, JP2-C15, JP2-C16, JP2-D15, JP2-D16, JP2-E15, JP2-E16, JP2-E17, JP2-F15, JP2-F16, JP2-F17
The following components are missing in the expected net JP2_JPU1_P3_LV_RETURN: JP2-A19, JP2-A20, JP2-A21, JP2-A22, JP2-E19, JP2-E20, JP2-E21, JP2-E22, JP2-F19, JP2-F20
The following components are missing in the expected net JP1_JPU0_P3_LV_SOURCE: JP1-B15, JP1-B16, JP1-C15, JP1-C16, JP1-D15, JP1-D16, JP1-E15, JP1-E16, JP1-E17, JP1-F15, JP1-F16, JP1-F17
The following components are missing in the expected net JP1_JPU0_P3_LV_RETURN: JP1-A19, JP1-A20, JP1-A21, JP1-A22, JP1-E19, JP1-E20, JP1-E21, JP1-E22, JP1-F19, JP1-F20
The following components are missing in the expected net JP1_JPU0_P2_EAST_LV_SOURCE: JP1-E1, JP1-E2, JP1-E3, JP1-E4, JP1-E5, JP1-E6, JP1-E7
The following components are missing in the expected net JP1_JPU0_P2_EAST_LV_RETURN: JP1-A6, JP1-B6, JP1-C6, JP1-C7, JP1-D6, JP1-D7, JP1-E11
[... cut ...] JP1-E27
The following components are missing in the expected net JP1_JPL0_P4_LV_RETURN: JP1-A30, JP1-B30, JP1-E30, JP1-E31, JP1-E32, JP1-E33
The following components are missing in the expected net JP0_JPU0_P3_LV_SOURCE: JP0-B15, JP0-B16, JP0-C15, JP0-C16, JP0-D15, JP0-D16, JP0-E15, JP0-E16, JP0-E17, JP0-F15, JP0-F16, JP0-F17
The following components are missing in the expected net JP0_JPU0_P3_LV_RETURN: JP0-A19, JP0-A20, JP0-A21, JP0-A22, JP0-E19, JP0-E20, JP0-E21, JP0-E22, JP0-F19, JP0-F20
```

ZishuoYang, 2019-05-03:
After getting an updated netlist (1May2019), and after updating our scripts to handle some corner cases, Yipeng and I found the following errors:

> ========0. JS connector not present in Pigtail power net========
> No JS connector found in JP3_JPU0_P2_WEST_LV_SOURCE
> No JS connector found in JP3_JPU0_P2_WEST_LV_RETURN
> No JS connector found in JP2_JPU0_P3_LV_SOURCE
> No JS connector found in JP2_JPU0_P3_LV_RETURN
> No JS connector found in JP1_JPL1_P4_LV_SOURCE
> No JS connector found in JP1_JPL1_P4_LV_RETURN
> No JS connector found in JP0_JPU0_P2_WEST_LV_SOURCE
> No JS connector found in JP0_JPU0_P2_WEST_LV_RETURN
> 
> ========2. Never used elinks========
> No biasing resistor found in JD9_FRO_ELK_P
> No biasing resistor found in JD9_FRO_ELK_N
> 
> ========3. Components missing========
> The following components are missing in the expected net JP2_JPU1_P3_LV_SOURCE: JP2-B15, JP2-B16, JP2-C15, JP2-C16, JP2-D15, JP2-D16, JP2-E15, JP2-E16, JP2-E17, JP2-F15, JP2-F16, JP2-F17
> The following components are missing in the expected net JP2_JPU1_P3_LV_RETURN: JP2-A19, JP2-A20, JP2-A21, JP2-A22, JP2-E19, JP2-E20, JP2-E21, JP2-E22, JP2-F19, JP2-F20
> The following components are missing in the expected net JP1_JPL0_P4_LV_SOURCE: JP1-A23, JP1-A24, JP1-A25, JP1-A26, JP1-B27, JP1-C27, JP1-D27, JP1-E27
> The following components are missing in the expected net JP1_JPL0_P4_LV_RETURN: JP1-A30, JP1-B30, JP1-E30, JP1-E31, JP1-E32, JP1-E33
> 

In translation, 
- Biasing resistor ref. des. to be corrected for JD9.

- `JP1_JPL1_P4_LV_(SOURCE|RETURN)` should be changed to JPL0:  
[image]

- `JP2_JPU0_P3_LV_(SOURCE|RETURN)` should be changed to JPU1:  
[image]

- On the JPU0 connector, need to change `JP(1|2)_JPU0_P2_WEST_(SOURCE|RETURN)` to JP(0|3). Similar to the previous P4 power swap, we can use my hybrid depop. table listed [here ](https://github.com/ZishuoYang/UT-Backplane-mapping/issues/59) for guidance.
[image]

- This error was found by eye -- should be `JP9_JPU2_P4_LV_RETURN`, not JP8:
[image]

- [ ] Check this after correction.