ZishuoYang/UT-Backplane-mapping #86: AsicToFiberMapping: Mirror BP's DCB indices need to be updated

ZishuoYang, 2019-12-08:
We see outdated DCB indices on Mirror BPs, while True BPs seem consistent. For example, on `Magnet-Bottom-C`, the DCB indices corresponding to pigtail `X-0-M` are shown to be 0/1/3, while we know the swapped indices on the BP schematics are 0/1/2.

yipengsun, 2019-12-16:
As discussed earlier, please take a look at https://github.com/umd-lhcb/pyUTM/blob/master/pyUTM/common.py, more specifically, check if `jd_swapping_mirror` is correct.

If that is correct, let's see if we can solve the problem either in-person or via vidyo after Tuesday.

ZishuoYang, 2019-12-19:
I think that the keys and values in jd_swapping_mirror are reversed. I will updated it and test run the mapping script.

ZishuoYang, 2019-12-19:
I have spot checked for consistency between updated AsicToFiberMapping in tag 0.15 and the Mirror BP schematics, on JP1,2,3,4,6,8,9. The new DCB indices are consistent.

closed 2019-12-21 by ZishuoYang