ZishuoYang/UT-Backplane-mapping #87: Additional netlist error-checking: depopulation resistors' reference designator flags

ZishuoYang, 2019-12-08:
We can add an additional automated check to the existing code, for depopulation resistors' reference designator flags, as described in the https://github.com/ZishuoYang/UT-Backplane-mapping/issues/18#issuecomment-461598415.

What Phoebe suggested on Slack, "no Rx should be attached to a line that ends up on a pigtail connector (ie is used in some variant) except for ground sense", is to get all the nets with a PT pin and check that there's no resistor present except for on ground sense lines. But this has to be done while identifying what nets should be active on a given variant. Because we are checking the netlist, we risk going in to a circle here to use the depopulation resistors for identifying active nets in the first place. 

I think the more practical way is to:
- make sure that the existing code to generate the depopulation auxiliary list is correct
- then inspect the netlist and compare the names of nets with depopulation resistors against the auxiliary list 

In the mean time, I have checked both True and Mirror schematics by eye. See details in the lab-notes issue post.

yipengsun, 2019-12-10:
Currently, the idea is the following:

1. To check `RB` mislabeled as `R`: List all elink and control nets in `JP0-7` that are not for reference only (`FRO`), and see if they contain a `R`. If they do, the `R`s are wrong
2. To check `RBSP` mislabeled as `RB`: List all elinks and control nets in `JP8-11` that has a `RB`, and see if the net involves **only** the 4th elinks of a hybrid. If not, the `RB`s are wrong.

yipengsun, 2019-12-11:
I manually changed the net:
```lisp
  (net "JD7_JP8_EC_RESET_GPIO_1_N"
    (node "C186" "2")
    (node "JP8" "i22")
    ; (node "RBSP_3" "1")  original
    (node "RB_3" "1")  ; manually changed to make sure the checker works
    (attr "DifferentialPair" "True" (textStyleRef "(Default)"))
  )
```

I can confirm that the `RBSP` mislabeled as `RB` is also effective:
```========0. Depopulation resistor labeling problem========
Incorrectly labeled resistor RB_3 found in JD7_JP8_EC_RESET_GPIO_1_N
Incorrectly labeled resistor R41 found in JD3_JP3_EC_HYB_i2C_SDA_1_P
Incorrectly labeled resistor R42 found in JD3_JP3_EC_HYB_i2C_SCL_1_P
Incorrectly labeled resistor R44 found in JD3_JP2_EC_RESET_GPIO_5_P
Incorrectly labeled resistor R43 found in JD3_JP2_EC_RESET_GPIO_3_P
```

yipengsun, 2019-12-11:
Finally, these are the real problems:
```========0. Depopulation resistor labeling problem========
Incorrectly labeled resistor R41 found in JD3_JP3_EC_HYB_i2C_SDA_1_P
Incorrectly labeled resistor R42 found in JD3_JP3_EC_HYB_i2C_SCL_1_P
Incorrectly labeled resistor R44 found in JD3_JP2_EC_RESET_GPIO_5_P
Incorrectly labeled resistor R43 found in JD3_JP2_EC_RESET_GPIO_3_P
```

ZishuoYang, 2019-12-11:
Great, the real ones listed above agree with my checking by eye, and you can find `RBSP` mislabeled as `RB`. There's one more thing: find `RBSP` mislabeled as `RSP`, since I found these shown in https://github.com/umd-lhcb/lab-notes/issues/55#issuecomment-563005216.

yipengsun, 2019-12-12:
I added a new rule to list all `RSP_\d+` resistors, on the assumptions that all these are mislabeled and require manual annotation. Here's the result:
```
Incorrectly labeled resistor RSP_5 found in JD11_JPL2_1V5
Incorrectly labeled resistor RSP_4 found in JD11_AGND
Incorrectly labeled resistor RSP_2 found in JD11_10_JPL2_2V5
Incorrectly labeled resistor RSP_3 found in JD10_JPL2_1V5
Incorrectly labeled resistor RSP_1 found in JD10_AGND
Incorrectly labeled resistor R41 found in JD3_JP3_EC_HYB_i2C_SDA_1_P
Incorrectly labeled resistor R42 found in JD3_JP3_EC_HYB_i2C_SCL_1_P
Incorrectly labeled resistor R44 found in JD3_JP2_EC_RESET_GPIO_5_P
Incorrectly labeled resistor R43 found in JD3_JP2_EC_RESET_GPIO_3_P
Incorrectly labeled resistor RSP_1 found in GND
```

ZishuoYang, 2019-12-12:
Great, I'll merge this and close the issue now.

closed 2019-12-12 by ZishuoYang

yipengsun, 2019-12-17:
These are the mirror backplane errors:
```
Incorrectly labeled resistor RSP_2 found in JD11_JPL2_1V5
Incorrectly labeled resistor RSP_1 found in JD11_AGND
Incorrectly labeled resistor RSP_4 found in JD11_10_JPL2_2V5
Incorrectly labeled resistor RSP_5 found in JD10_JPL2_1V5
Incorrectly labeled resistor RSP_3 found in JD10_AGND
Incorrectly labeled resistor R95 found in JD2_JP1_EC_HYB_i2C_SDA_1_P
Incorrectly labeled resistor R96 found in JD2_JP1_EC_HYB_i2C_SCL_1_P
Incorrectly labeled resistor R98 found in JD2_JP0_EC_RESET_GPIO_5_P
Incorrectly labeled resistor R97 found in JD2_JP0_EC_RESET_GPIO_3_P
Incorrectly labeled resistor RSP_1, RSP_2, RSP_3, RSP_4, RSP_5 found in GND
```

PR #88 (2019-12-12): Mislabeled resistor error checker
Also include a parser for ASIC to hybrid mapping to list `JD` slots usage for each `JP`.

This should fix #87.